hideakitai/ESP32SPISlave #1: Weird Behavior of Slave mode

openedhardware, 2021-07-28:
Hi, @hideakitai !

I am using the slave-echo example to send & receive the same data from the master to slave.

1. The slave library receives 2~3 times though master sent one time only.
2. It works sometimes, but most likely doesn't work.. seems timing issue?

    Here is a screenshot of master:
    
[image]

And slave (Received twice):
[image]

 Note that I am using 500kHz for the SCK and MODE3 on the master.(actually a Raspberry Pi).

Here is a sample code:
```python
import spidev
import RPi.GPIO as GPIO

spi = spidev.SpiDev()
spi.open(0, 0)
spi.max_speed_hz = 500000
spi.mode = 3

def _read_spi(index, cmd):
    GPIO.output(CS_PINS[index], False)
    spi.xfer(copy.deepcopy(cmd))
    val = spi.readbytes(BLOCK_SIZE)
    GPIO.output(CS_PINS[index], True)
    return val

req = [v for v in range(32)]

for i in range(10):
    reply = _read_spi(index=0, cmd=req)
    print('Sent: ', req, 'Reply: ', ' '.join([str(t) for t in reply]))
    req = req[1:] + [req[0]]
    time.sleep(1)
```

hideakitai, 2021-07-28:
I think the master's gpio should be reverted to be compatible with `slave_echo` example? The initial state should be high and start the transaction at a falling edge (high -> low). The transaction ends with a rising edge (low -> high).

openedhardware, 2021-07-28:
Yeah, starts from `GPIO.output(CS_PINS[index], False)` and ends with `GPIO.output(CS_PINS[index], True)`... lol

hideakitai, 2021-07-28:
I don't know how the spi works with RPi.GPIO, but I think this line transfers bytes (and you are not using returned bytes from slave)

```Python
    spi.xfer(copy.deepcopy(cmd))
```

And also this line transfers again

```Python
    val = spi.readbytes(BLOCK_SIZE)
````

Maybe you should just use

```Python
    val = spi.xfer(copy.deepcpy(cmd))
```

hideakitai, 2021-07-28:
In addition to it, you should use `xfer2([val])` maybe. `xfer([val])` makes CS inactive (high) per byte, but `xfer2([val])` keeps active (low) until all bytes are sent.

Where is the detailed documentation of RPi.GPIO...

openedhardware, 2021-07-28:
You rock!

Yeah, `xfer2()` function works like a charm! But there is another issue.. lol

Seems like SPI Slave replies previous buffer data instead of the current one?

Please check this log:
```
Sent:  [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31] Reply:  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3 4 5 6 7 8
Sent:  [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0] Reply:  0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31
Sent:  [2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1] Reply:  1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0
Sent:  [3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2] Reply:  2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1
Sent:  [4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3] Reply:  3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2
Sent:  [5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3, 4] Reply:  4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3
Sent:  [6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3, 4, 5] Reply:  5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3 4
Sent:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3, 4, 5, 6] Reply:  6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3 4 5
Sent:  [8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3, 4, 5, 6, 7] Reply:  7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3 4 5 6
Sent:  [9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 0, 1, 2, 3, 4, 5, 6, 7, 8] Reply:  8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 0 1 2 3 4 5 6 7
```

hideakitai, 2021-07-28:
It's correct SPI behavior. See
https://learn.sparkfun.com/tutorials/serial-peripheral-interface-spi/all
https://www.engineersgarage.com/spi-what-is-serial-peripheral-interface-protocol/

hideakitai, 2021-07-30:
@openedhardware I think the library is working fine, so can I close this issue?

hideakitai, 2021-08-02:
@openedhardware If you still have any problems, please let me know. Closing.

closed 2021-08-02 by hideakitai