hideakitai/ESP32SPISlave #4: TX data issue

Fasol0, 2021-11-03:
Hi,

Where should the spi_slave_tx_buf be filled, or the proper way to send back data to the master?

If I don't touch tx_buf, I get consistent rx data from master on rx_buf, but if I change one byte in the tx_buf, it overwrites the rx_buf, wherever I set the tx_buf. 

Is it a bug or am I doing something wrong?

Best regards

hideakitai, 2021-11-04:
@Fasol0 Hi,

`master_slave_polling` example shows how to set tx_buf. Does this example work fine?

https://github.com/hideakitai/ESP32SPISlave/blob/2df86e38a803899176ed99fcbc73780a012f208a/examples/master_slave_polling/master_slave_polling.ino#L136

Fasol0, 2021-11-04:
I use the slave_simple_polling example exactly as it is. 
I just replaced HSPI to VSPI. 

If I comment set_buffer() I get correct rx data.
If i uncomment it, I get 0 for all rx_buf bytes, except rx_buf[0] which seems to be a random value

I don't understand how setting tx_buf in setup can interfere with the rx_buf,

do you have any idea?

hideakitai, 2021-11-05:
If you comment out `set_buffer()` in `setup()`, `tx_buf` won't be initialized so it would be uninitialized data. You can change `tx_buf` before queuing the transaction like:

```c++
    // if there is no transaction in queue, add transaction
    if (slave.remained() == 0) {

        // change tx_buf here

        slave.queue(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);
    }
```

Is this what you want to know?

Fasol0, 2021-11-05:
Here is the thing: If I initialize TX buffer then I don't receive proper RX data..

`void set_buffer()
{
    // for (uint32_t i = 0; i < BUFFER_SIZE; i++) {
    //      spi_slave_tx_buf[i] = (0xFF - i) & 0xFF;
    // }
     memset(spi_slave_rx_buf, 0, BUFFER_SIZE);
}

void spi_init() 
{
  slave.setDataMode(SPI_MODE3);
  slave.setQueueSize(1);  // transaction queue size
  slave.begin(VSPI);

  set_buffer();
}

get_spi_data()
{
    if (slave.remained() == 0)
    {
        slave.queue(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);
    }
    while (slave.available()) 
    {   
        for (size_t i = 0; i < BUFFER_SIZE; ++i)
            printf("%d ", spi_slave_rx_buf[i]);
        printf("\n");

        slave.pop();
    }
}`

spi_init() is called inside setup() , and get_spi_data() inside loop().
If tx buffer is not initialized (commented), I read proper rx buffer from my master (which is a 8 byte frame,  "11 21 31 41 51 61 71 81")
If i do uncomment tx buffer initialization, I get  "255 0 0 0 0 0 0 0" as rx buffer

Fasol0, 2021-11-05:
Ok this might help:

I tested the master_slave_polling example as you suggested
And I get "[WARNING] queue is full with transactions. discard new transaction request" everytime.

Fasol0, 2021-11-08:
[image]

Blue signal is SCLK
Red signal is MOSI (slave RX)
Green signal is MISO (slave TX)
Last one is CS

On this example I use slave_simple_polling just as it is. I only initialize tx_buf, I never change its values anywhere else,
and yet you can see that rx buffer (data from red signal) as an effect on the tx buffer.

Therefore it does not seem possible to use this library to transmit data from slave to master for  now..

hideakitai, 2021-11-08:
Ok, I will check it tomorrow, so could you summarize followings?

- esp32 version
- what you want to do
- what is the problem
- minimum code that can reproduce the issue

Fasol0, 2021-11-08:
Ok thank you,

- I have tested the code on several esp32 based boards, all have the same issue (WEMOS lolin D32, ESP32-WROOM32-UE, etc..)
- I just want to send back data to master
- If spi_slave_tx_buf is initalized, rx_buf AND tx_buf data are messed up (random values, mostly zeros),
- simple_slave_polling example does not work as is (no changes): to get proper rx data, I need to comment spi_slave_buffer initialization. I use VSPI as SPI pins.

This post seems really close to the issue:
https://esp32.com/viewtopic.php?t=11655

unfortunately his solution (turning on CONFIG_SPI_SLAVE_IN_IRAM) is not possible since we don't have access to sdk_config (Arduino framework using compiled version of config files)

My platformio.ini:
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200

hideakitai, 2021-11-08:
Thanks, I want more information to reproduce the problem.

> esp32 version
- esp32dev in PlatformIO

Which version? Latest one is v3.3.2.

> what you want to do
- just send back data to master

Ok so you need to set the received bytes to tx_buf

> what is the problem
- If spi_slave_tx_buf is initalized, rx_buf AND tx_buf data are messed up (random values, mostly zeros),

I see

> minimum code that can reproduce the issue

- simple_slave_polling example does not work as is (no changes): to get proper rx data, I need to comment spi_slave_buffer initialization. I use VSPI as SPI pins.

Where is SPI master? Another device? I want whole code that can reproduce the problem.

hideakitai, 2021-11-09:
In `master_slave_polling` and `master_slave_polling_large_bytes` examples, `memset` inside of `set_buffer` was not needed. Because of `memset`, buffers are cleared to zero. So the transaction results were always zero. However, the library itself worked fine.

I've fixed examples in v0.1.2. Is this release fix your issue?

hideakitai, 2021-11-09:
I'm sorry, I must have misunderstood because it doesn't seem to work correctly with ESP32 v1.0.5 or later in the Arduino IDE. I don't have time to look into it right now, but I'll do so later.

As of now, if you use ESP32DMASPI with ESP32 v1.0.4, it should work with Slave.

Fasol0, 2021-11-09:
Hi,

I tried ESP32DMASPISlave, it does not work either, weither I try with v1.0.6 or v1.0.4
In this case RX buffer never gets updated. I always get zeros (value from initialization) or 254 if I comment memset()

Fasol0, 2021-11-22:
Hi,

Do you think you would have time to dig into that issue in the next few days?
In the other case I will implement another communication protocol for my application.

Regards

hideakitai, 2022-01-02:
Sorry for the late reply. I'm going to archive this repo and merge into ESP32DMASPI.

closed 2022-01-02 by hideakitai