hideakitai/ESP32SPISlave #5: simple way to do LSB transmissions ?

L10N37, 2021-12-06:
I am using your library for a project but it's a console that sends LSB first.

Is there a simple way to change this as I'm having to convert everything at the moment, just making baby steps but this is a big issue for a beginner :P

hideakitai, 2022-01-02:
Sorry for the late reply. Selecting MSB/LSB is the scope of master device. And most of the devices use MSB first by default, so I don't have a plan to do it. Please refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#transactions-with-integers-other-than-uint8-t

closed 2022-01-02 by hideakitai

L10N37, 2022-01-02:
someone helped me fix this a few weeks back. line30 of slave CPP. 

if_cfg.flags= SPI_SLAVE_BIT_LSBFIRST; 

it seems to have worked as I've captured a lot of successful transmissions.