hideakitai/ESP32SPISlave #6: Latest version missing CPP files

L10N37, 2022-01-03:
no CPP files?

hideakitai, 2022-01-03:
Yes, changed to header only. Failed to compile?

L10N37, 2022-01-03:
I thought that might be the case, fails yes.

``
Compiling .pio/build/esp32dev/FrameworkArduino/FunctionalInterrupt.cpp.o
Archiving .pio/build/esp32dev/lib10d/libSPI.a
Indexing .pio/build/esp32dev/lib10d/libSPI.a
Compiling .pio/build/esp32dev/FrameworkArduino/HardwareSerial.cpp.o
In file included from src/main.cpp:2:0:
include/ESP32SPISlave.h:48:28: error: 'SOC_SPI_MAXIMUM_BUFFER_SIZE' was not declared in this scope
         .max_transfer_sz = SOC_SPI_MAXIMUM_BUFFER_SIZE,
                            ^
include/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
include/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
include/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
include/ESP32SPISlave.h:174:69: error: 'SPI_DMA_DISABLED' was not declared in this scope
         esp_err_t e = spi_slave_initialize(host, &bus_cfg, &if_cfg, SPI_DMA_DISABLED);
                                                                     ^
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1

`

hideakitai, 2022-01-03:
What is the version of esp firmware? Please try the latest one 2.0.2 maybe

hideakitai, 2022-01-03:
Maybe 3.4.0 if you use platformio

hideakitai, 2022-01-06:
@L10N37 v0.2.0 requires ESP32 version >= 2.0.0 for Arduino IDE, and not supported on PlatformIO now. Added supported version matrix to README

L10N37, 2022-01-06:
I'm using platform IO... so I reverted to old version

hideakitai, 2022-01-06:
Someday platform-esp will support a newer version of esp-idf!!