hideakitai/ESP32SPISlave #7: 18 different 16 Bit transactions,

Vani456, 2022-04-12:
Hi,

I'm trying to transfer 18 different transactions with 16 bit each from slave to master, the Clock is set to 125kHz.

Therefor I put the CC on another free GPIO pin and attached an interrupt on this additional "CS Line". In the ISR I check for the rising edge and then write the data from an array to spi_slave_tx_buf[0]. Because I want 18 different data to be transmitted (for testing purposes, 0xAAAA and 0xFFFF should be transmitted alternately) I use a counter: 

>  if (enableLOW == false && DirectionSignal == LOW){
      spi_slave_tx_buf[0] = TMData[counter];
      ++counter;
      if(counter > 18) {
        counter = 0;
      }      
  }

The DirectionSignal is just a Signal with the value 0/1, on this Signal depends if the slave shall send or receive data.

If the ISR is done and I went back to my code there is just this code to transmit the data:

> if (DirectionSignal == LOW) {
          if (slave.remained() == 0) {
              slave.queue(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);
          }
  }
while (slave.available()) {
  slave.pop();
}

When I check the transmitted data i receive the following bit patterns:

[image]

I've already tried it with 8Bit and it was working as expected. 

Since my board version is incompatible with the latest version of your library, I use the previous version 0.1.2. There I changed the parameters(rx_buf, tx_buf) of the functions from uint8_t to uint16_t. But since it still does not work, I have the impression that I overlook something at one point that still needs to be changed.

I hope you can help me with this!
Vani456

tarzanboy13, 2022-04-19:
I have same issue, and using newest lib version...  Trying to recv/send 32 Bits

hideakitai, 2023-10-05:
Could you try it with the latest arduino-esp32 core (not the library)?

hideakitai, 2023-10-05:
If you still need help with the latest arduino-esp32 core and the latest ESP32SPISlave, please feel free to reopen.

In that case, please attach a minimal code that can reproduce the example.

closed 2023-10-05 by hideakitai