hideakitai/ESP32SPISlave #8: HSPI_HOST, VSPI_HOST and VSPI not declared

dalymople, 2022-05-14:
I am trying to compile a program for a ESP32 C3 in Arduino IDE and I am getting compile errors that `HSPI_HOST`, `VSPI_HOST` and `VSPI` are not defined.

It looks like the HAL headers for the C3 use `SPI2_HOST` and `SPI3_HOST` instead, I can't work out the alternative definition for VSPI, but I am trying to use HSPI so I just defined it not to conflict. Defining these before including the library seems to resolve the error, but I haven't tested with hardware yet.

zzzbatmand, 2022-11-21:
In the ESP32c3 file "spi_types.h" I found the following:
``` c
typedef enum {
//SPI1 can be used as GPSPI only on ESP32
    SPI1_HOST=0,    ///< SPI1
    SPI2_HOST=1,    ///< SPI2
    SPI3_HOST=2,    ///< SPI3
} spi_host_device_t;
```

This matches the deffinitions in "spi_common.h" from an older version of the ESP-IDF (V3.3):
``` c
typedef enum {
    SPI_HOST=0,                     ///< SPI1, SPI
    HSPI_HOST=1,                    ///< SPI2, HSPI
    VSPI_HOST=2                     ///< SPI3, VSPI
} spi_host_device_t;
```

I haven't found a reference to "VSPI" but it looks like it should be the same as "VSPI_HOST"
If I define these before calling "#include <ESP32SPISlave.h>" it compiles just fine, but goes in a boot loop, once I try to run the actual code.

zzzbatmand, 2022-11-22:
I have fixed the issue and made a pull request: #12

Lenbus, 2023-04-20:
That resolved my issues in PlatformIO as well. 👍 
Somehow these changes don't seem to be merged yet, so I manually had to alter the ESP32SPISlave.h file... 🤔

hideakitai, 2023-08-13:
Fixed by https://github.com/hideakitai/ESP32SPISlave/pull/21

closed 2023-08-13 by hideakitai