hideakitai/ESP32SPISlave #9: Invalid reception

monstermichl, 2022-09-18:
Hi hideakitai,

thanks for providing a dedicated SPI slave library for the ESP32. I would like to transfer SPI data from one ESP32 to another. However, the data received is most of the time totally different then the sent one. As a test I'm only sending 1, 2, 3, 4 per transaction. So the receive buffer is 4 bytes long but it doesn't really matter how long the buffer is, the result is always similar. The oscilloscope is showing me that the data is sent correctly. I also tried to send with a lower frequency and played around with modes but that didn't help either. Any idea on that? Thanks in advance.

[image]

Sender:
```cpp
#include <Arduino.h>
#include <SPI.h>
#include "config.h"

/* ----------------- Defines ------------------------------- */
#define CS_PIN (15u)

/* ----------------- Variables ----------------------------- */

/* ----------------- Function declarations ----------------- */
static void spiSetup(void);
static void spiWrite(uint16_t channel1, uint16_t channel2);

/* ----------------- Arduino specific functions ------------ */
void setup()
{
    Serial.begin(9600);

    spiSetup();
}

void loop()
{
    spiWrite(0, 0);
}

/* ----------------- Function implementations -------------- */
static void spiSetup(void)
{
    pinMode(CS_PIN, OUTPUT);
    digitalWrite(CS_PIN, LOW);

    SPI.begin();
}

static void spiWrite(uint16_t channel1, uint16_t channel2)
{
    SPI.beginTransaction(SPISettings(1000000, MSBFIRST, SPI_MODE0));
    digitalWrite(CS_PIN, LOW);
    SPI.transfer(1);
    SPI.transfer(2);
    SPI.transfer(3);
    SPI.transfer(4);
    digitalWrite(CS_PIN, HIGH);
    SPI.endTransaction();
}
```

Receiver:
```cpp
#include <Arduino.h>
#include "ESP32SPISlave.h"
#include "config.h"

/* ----------------- Defines ------------------------------- */
#define SPI_BUFFER_SIZE (4u)

/* ----------------- Variables ----------------------------- */
ESP32SPISlave slave;
uint8_t spiTxBuffer[SPI_BUFFER_SIZE];
uint8_t spiRxBuffer[SPI_BUFFER_SIZE];

/* ----------------- Function declarations ----------------- */
static void setupSpi(void);

/* ----------------- Arduino specific functions ------------ */
void setup()
{
    Serial.begin(9600);

    setupSpi();
}

void loop()
{
    slave.wait(spiRxBuffer, spiTxBuffer, SPI_BUFFER_SIZE);

    if (slave.available())
    {
        for (size_t i = 0; i < SPI_BUFFER_SIZE; ++i) {
            printf("%d ", spiRxBuffer[i]);
        }
        printf("\n");

        slave.pop();
    }
}

/* ----------------- Function implementations -------------- */
static void setupSpi(void)
{
    slave.setDataMode(SPI_MODE0);
    slave.begin(VSPI);
}
```

hideakitai, 2023-10-05:
I think it may be caused by hardware noise, but do you mean it is not?

> The oscilloscope is showing me that the data is sent correctly.

Could you try it again with the latest arduino-esp32 core (not the library)?

hideakitai, 2023-10-05:
If you still need help with the latest arduino-esp32 core and the latest ESP32SPISlave, please feel free to reopen.

closed 2023-10-05 by hideakitai