hideakitai/ESP32SPISlave #10: Compile errors

PribaNosati, 2022-10-15:
Trying to compile your first example with target "ESP32 Dev Module" and using Arduino IDE 2.0.0, I get error messages.
Sketch:
#include <ESP32SPISlave.h>

ESP32SPISlave slave;

static constexpr uint32_t BUFFER_SIZE {32};
uint8_t spi_slave_tx_buf[BUFFER_SIZE];
uint8_t spi_slave_rx_buf[BUFFER_SIZE];

void setup() {
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12 -> default
    // VSPI = CS:  5, CLK: 18, MOSI: 23, MISO: 19
    slave.setDataMode(SPI_MODE0);
    slave.begin(HSPI);
}

void loop() {
    // block until the transaction comes from master
    slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);

    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and `spi_slave_rx_buf` is automatically updated
    while (slave.available()) {
        // do something with `spi_slave_rx_buf`

        slave.pop();
    }
}

Errors:

In file included from C:\Users\priba\AppData\Local\Temp\.arduinoIDE-unsaved2022914-5736-1o67564.9opq\sketch_oct14a\sketch_oct14a.ino:1:0:
C:\Users\priba\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:48:28: error: 'SOC_SPI_MAXIMUM_BUFFER_SIZE' was not declared in this scope
         .max_transfer_sz = SOC_SPI_MAXIMUM_BUFFER_SIZE,
                            ^
C:\Users\priba\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\Users\priba\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\priba\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
C:\Users\priba\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:174:69: error: 'SPI_DMA_DISABLED' was not declared in this scope
         esp_err_t e = spi_slave_initialize(host, &bus_cfg, &if_cfg, SPI_DMA_DISABLED);
                                                                     ^

exit status 1

Compilation error: exit status 1

anamariarincon03, 2023-01-10:
SAME ERROR.

I am trying to compile an ESP32 Slave Sketch using ESP32SPISlave Library. I am compiling for "AI Thinker ESP32-CAM" and using Arduino IDE 2.0.3. I get error messages too and I don't know how to fix it.

I got the slave sketch from here: https://microcontrollerslab.com/esp32-spi-communication-tutorial-arduino/?unapproved=241787&moderation-hash=b99233cd69eff588e2c234e26dc05aa3#comment-241787

Sketch: 
#include <ESP32SPISlave.h>

ESP32SPISlave slave;

static constexpr uint32_t BUFFER_SIZE {32};
uint8_t spi_slave_tx_buf[BUFFER_SIZE];
uint8_t spi_slave_rx_buf[BUFFER_SIZE];

#define LED 2
void setup() {
    Serial.begin(115200);
    delay(2000);
    pinMode(LED, OUTPUT);
    // begin() after setting
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12 -> default
    // VSPI = CS:  5, CLK: 18, MOSI: 23, MISO: 19
    slave.setDataMode(SPI_MODE0);
    slave.begin();
    // slave.begin(VSPI);   // you can use VSPI like this

    // clear buffers
    memset(spi_slave_tx_buf, 0, BUFFER_SIZE);
    memset(spi_slave_rx_buf, 0, BUFFER_SIZE);
}

void loop() {
    // block until the transaction comes from master
    slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);

    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated
    char data;
    while (slave.available()) {
        // show received data
         Serial.print("Command Received: ");
         Serial.println(spi_slave_rx_buf[0]);
         data = spi_slave_rx_buf[0];
         slave.pop();
    }
    if(data == 1 )
    {
        Serial.println("Setting LED active HIGH ");
        digitalWrite(LED, HIGH);
    }
    else if(data == 0 )
    {
        Serial.println("Setting LED active LOW ");
        digitalWrite(LED, LOW);
    }
     Serial.println("");
}

I get error messages as follows:

In file included from C:\Users\Ana Maria\Documents\Arduino\TakePhoto_SLAVE_esp32cam\TakePhoto_SLAVE_esp32cam.ino:1:0:
C:\Users\Ana Maria\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:48:28: error: 'SOC_SPI_MAXIMUM_BUFFER_SIZE' was not declared in this scope
         .max_transfer_sz = SOC_SPI_MAXIMUM_BUFFER_SIZE,
                            ^
C:\Users\Ana Maria\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\Users\Ana Maria\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Ana Maria\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
C:\Users\Ana Maria\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:174:69: error: 'SPI_DMA_DISABLED' was not declared in this scope
         esp_err_t e = spi_slave_initialize(host, &bus_cfg, &if_cfg, SPI_DMA_DISABLED);
                                                                     ^

exit status 1

Compilation error: exit status 1

PribaNosati, 2023-01-10:
Sorry, I do not know much about ESP32 libraries. You should ask 
library's author.

cmtk, 2023-01-30:
I am seeing this issue as well with the same two errors.

I am using an ESP32 devkit-v1 board, which has an ESP-WROOM-32 module/esp32 soc.

victor-cortez, 2023-02-06:
I also am getting the samme errors you are getting while using the latest version of the library.
I am compiling for the firebeetle-esp32 using the arduino IDE.

hideakitai, 2023-08-01:
This library does not support the latest ESP32 core now. Please let me know:

- ESP32 Board (esp32, s3, c3, etc): 
- ESP32 Board Firmware Version (see board manager):
- Library Version of ESP32SPISlave:

hideakitai, 2023-08-13:
Fixed by https://github.com/hideakitai/ESP32SPISlave/pull/21

closed 2023-08-13 by hideakitai