hideakitai/ESP32SPISlave #13: Compilation error: driver/spi_slave.h

lezlikov, 2022-11-25:
Compilation error: driver/spi_slave.h: No such file or directory
Help pls!

lezlikov, 2022-11-25:
[image]

AndyW999, 2023-04-06:
I also get this -

Compilation error: driver/spi_slave.h: No such file or directory

Using Arduino 2.0.4

And if I try and use PlatformIO as soon as it goes into the Loop - 

E (154) spi_slave: spi_slave_queue_trans(291): data transfer > host maximum

Any ideas anybody?

Thanks

Andy.

hideakitai, 2023-08-01:
This library does not support the latest ESP32 core now. Please let me know:

- ESP32 Board (esp32, s3, c3, etc): 
- ESP32 Board Firmware Version (see board manager):
- Library Version of ESP32SPISlave:

hideakitai, 2023-08-13:
Fixed by https://github.com/hideakitai/ESP32SPISlave/pull/21

closed 2023-08-13 by hideakitai