hideakitai/ESP32SPISlave #18: `available` unclear

Erhannis, 2023-07-11:
What is `available`, exactly?  From the readme: "available() returns size of results of transaction", I thought it gave the number of bytes sent by the other side.  Nope.  I'm getting "1" from `available`, but if I check the buffer, I see the 32 bytes I sent from the master.  How do you know how many bytes the master sent?  What does `available` actually tell you?

hideakitai, 2023-08-01:
Please take a look at the README.

```cpp
// transaction result info
size_t available() const;  // size of completed (received) transaction results
size_t remained() const;   // size of queued (not completed) transactions
uint32_t size() const;     // size of the received bytes of the oldest queued transaction result
void pop();                // pop the oldest transaction result
```

closed 2023-08-01 by hideakitai

Erhannis, 2023-08-02:
I did read the readme; I quoted the readme in my question.  Did you read my question?  The code isn't doing what the readme says it should.  I send 32 bytes from master, get "1 available" on the slave side, yet the slave's buffer indeed contains all 32 bytes sent from master.

Erhannis, 2023-08-02:
Unless "size of transaction results" means e.g. "number of transactions" instead of "number of bytes in transaction"?

Erhannis, 2023-08-02:
Oh wait - the explanation for "size()" hints that the latter explanation is indeed what you mean.  If that IS the case, might I recommend changing "size of completed (received) transaction results" into "number of completed (received) transactions"?  Otherwise it sounds like "size of the results" = "byte count".

hideakitai, 2023-08-02:
If you think so, I'm looking forward to your pull request :)

Erhannis, 2023-08-02:
Here you go; #19 ; thanks!

hideakitai, 2023-08-02:
Merged, thanks!