hideakitai/ESP32SPISlave #24: ESP32 as SPI Slave trying to receive 12 bit transfers.

ValdezNC, 2023-11-17:
I am trying to use ESP32SPISlave to intercept the data between keypad and mostly passive LCD display with icons. Using an digital analyzer it uses what is very similar to SPI to change the LCD display, but it sends 12 bits when the CS is pulled low. The device sends 18 x 12 bit bytes to the screen and the last four bits are the ones that change the state of the icon. I can remove the display and manually control it using by manually settign the SS low and transfering the data, however when I try to listen in as a slave I only can see the first 8 bits. I have tried looking at the next data in the buffer array i.e. spi_slave_rx_buf[1], spi_slave_rx_buf[2]... but the data is not there. 

Is there any way to modify my code or the ESP32SPISLave.h file to see the full 12 bits or extract the last 4 bits seperately?

The code I am using is:

```cpp
#include <ESP32SPISlave.h>

ESP32SPISlave slave;

static constexpr uint32_t BUFFER_SIZE {32};
uint8_t spi_slave_tx_buf[BUFFER_SIZE];
uint8_t spi_slave_rx_buf[BUFFER_SIZE];

void setup() {
    Serial.begin(115200);
    // begin() after setting
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12 -> default
    // VSPI = CS:  5, CLK: 18, MOSI: 23, MISO: 19
    slave.setDataMode(SPI_MODE3);
    slave.begin();
    // slave.begin(VSPI);   // you can use VSPI like this

    // clear buffers
    memset(spi_slave_tx_buf, 0, BUFFER_SIZE);
    memset(spi_slave_rx_buf, 0, BUFFER_SIZE);
}

void loop() {
    // block until the transaction comes from master
    slave.wait(spi_slave_rx_buf, BUFFER_SIZE);

    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated
    char data;
    while (slave.available()) {
        // show received data
         Serial.println(spi_slave_rx_buf[0], HEX);
         data = spi_slave_rx_buf[0];
         slave.pop();
    }

}
```

Analyzer signal:
[image]

Serial Output:
[image]

hideakitai, 2023-11-17:
Does your LCD become an SPI master? Or does the status data come from LCD as a Slave after you send data as a Master?

ValdezNC, 2023-11-18:
I believe the LCD is a SPI slave and the keypad is the master. I am hooking my ESP32 into the 3 data lines (clk, ss, mosi) running from the keypad to display to sniff the data as it's being sent from keypad to lcd. So i assumed my esp32 would also act like a slave in parrallel with the LCD and receive the same data.

hideakitai, 2023-11-18:
In your code, you are printing only the first byte per transaction. Is this correct?

```cpp
    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated
    char data;
    while (slave.available()) {
        // show received data
         Serial.println(spi_slave_rx_buf[0], HEX);
         data = spi_slave_rx_buf[0];
         slave.pop();
    }
```

If not, please refer to my example:

https://github.com/hideakitai/ESP32SPISlave/blob/ec346e2b50b3611e7fe8213c65f12cd810789669/examples/slave_simple_wait/slave_simple_wait.ino#L29-L41

ValdezNC, 2023-11-19:
Sorry, I did try looping through the spi_slave_rxbuf[i] array for each transaction. 
Only spi_slave_rx_buf[0] is correct and matches the first 8 bit bytes from the logic analyzer. (i.e. 0x9c, 0x80, 0x83, 0x85...) But second byte in the array or spi_slave_rx_buf[1] is the same for every transaction i.e. 0xF7 and incorrect. It should be 0x7, 0xB, 0x0, 0x3...). Even if it inputted 0s for the last 4 bits (0x70, 0xB0, 0x00, 0x30...) I could work with that but it gives me the same constant result for every transaction 0xF7 (when i get back to my bench I will hook it up again and screenshot the data). 

It is like the second byte does not get recorded at all because only 4 of the 8 bits of the second byte are transmitted before the CS goes high and the next transaction starts. If I could make it read 4 bit bytes at a time instead of 8 per transaction, all the data should make it into spi_slave_rx_buf[i] evenly. So for the data would come out:

1st transaction:
spi_slave_rx_buf[0] = 0x9, spi_slave_rx_buf[1] = 0xC, spi_slave_rx_buf[2] = 0x7.
2nd transaction:
spi_slave_rx_buf[0] = 0x8, spi_slave_rx_buf[1] = 0x0, spi_slave_rx_buf[2] = 0x3.
3rd transaction:
spi_slave_rx_buf[0] = 0x8, spi_slave_rx_buf[1] = 0x0, spi_slave_rx_buf[2] = 0xB.
...

hideakitai, 2023-12-10:
Could you give me code that can reproduce the same issue in my environment, with one of the SPIs on the ESP32 as a dummy Master (sending the same byte sequence as the target) and another SPI on the same ESP32 as a Slave (causing the same phenomenon as the issue)?

ValdezNC, 2023-12-12:
I don't know if I'll be able to write code to mimmick that hardware. The signal is being sent from an alarm keypad that I'm attempting to intercept the lcd display data to read the alarm status remotely.

However I got a different keypad model that uses a standard HD44780 16x2 display and I was able to decode and read the data correctly on that model and so haven't worked on the other one that uses the odd 12 bit SPI signal in a while. Ill take another look at the 12 bit SPI keypad again soon and see if I can provide better details.

hideakitai, 2024-01-04:
If you still need help, please feel free to reopen this issue.

closed 2024-01-04 by hideakitai