hideakitai/ESP32SPISlave #25: ESP32-S3 as Slave, changes tft background colour to blue

navycuda, 2023-11-25:
I have an Adafruit Qualia ESP32-S3 RGB666 board that is being used to drive a 40p 320x820px tft display.  I'm using a feather CAN M4 to listen to a few different sensors and the CAN bus.  That feather then will transmit the results to the Qualia for display.

As soon as slave.begin() is called the background colour turns a light blue.  If I hold the BOOT0 button on the qualia board the blue backlighting goes away.  The combination otherwise appears to work correctly.  If I comment out _slave.begin(), then the display does not go through a background colour change.

I have tried a number of different delays, having the display and the spi slave as global vs being in a class.  Where should I be looking to figure out why this is?

Display Initialization:
``` cpp
  Arduino_XCA9554SWSPI    * _expander;
  Arduino_ESP32RGBPanel   * _rgbpanel;
  Arduino_RGB_Display     * _gfx;

  void InitializeDisplay()
  {
    _expander = new Arduino_XCA9554SWSPI(
      PCA_TFT_RESET, PCA_TFT_CS, PCA_TFT_SCK, PCA_TFT_MOSI,
      &Wire, 0x3F);

    _rgbpanel = new Arduino_ESP32RGBPanel(
      TFT_DE, TFT_VSYNC, TFT_HSYNC, TFT_PCLK,
      TFT_R1, TFT_R2, TFT_R3, TFT_R4, TFT_R5,
      TFT_G0, TFT_G1, TFT_G2, TFT_G3, TFT_G4, TFT_G5,
      TFT_B1, TFT_B2, TFT_B3, TFT_B4, TFT_B5,
      1 /* hsync_polarity */, 50 /* hsync_front_porch */, 2 /* hsync_pulse_width */, 44 /* hsync_back_porch */,
      1 /* vsync_polarity */, 16 /* vsync_front_porch */, 2 /* vsync_pulse_width */, 18 /* vsync_back_porch */
      );

    _gfx = new Arduino_RGB_Display(
      // 3.2" 320x820 rectangle bar display
      320 /* width */, 820 /* height */, _rgbpanel, 1 /* rotation */, true /* auto_flush */,
      _expander, GFX_NOT_DEFINED /* RST */, tl032fwv01_init_operations, sizeof(tl032fwv01_init_operations));

  #ifdef GFX_EXTRA_PRE_INIT
    GFX_EXTRA_PRE_INIT();
  #endif
    Wire.setClock(1000000); // speed up I2C 
    if (!_gfx->begin()) {
      Serial.println("gfx->begin() failed!");
    }

    // Init Display
    _gfx->fillScreen(BLACK);
    delay(100);
  };
```

SPISlave Initialization:
``` cpp
  SpiCommunications::SpiCommunications()
  {
    _slave.setDataMode(SPI_MODE0);
    _slave.begin(HSPI);
  };
```

hideakitai, 2023-12-10:
Could you give me code that can reproduce the same issue in my environment, with one of the SPIs on the ESP32 as a dummy Master (sending the same byte sequence as the target) and another SPI on the same ESP32 as a Slave (causing the same phenomenon as the issue)?

navycuda, 2023-12-10:
> Could you give me code that can reproduce the same issue in my environment, with one of the SPIs on the ESP32 as a dummy Master (sending the same byte sequence as the target) and another SPI on the same ESP32 as a Slave (causing the same phenomenon as the issue)?

Please forgive, I'm still rather new at this.  I have seemingly isolated it to the instantiation order:

```c++
void setup(void)
{ 
  Serial.begin(115200);
  slave.setDataMode(SPI_MODE0);
  slave.begin(HSPI);

  logic = new Controllers::Logic();

  logic->SplashScreen(1500);
  logic->ViewCluster();
};
````

Does not produce the issue, however if I begin the slave after logic it does.

The display instantiation is now buried into logic but is otherwise the same as the op.

Do you want my whole project? I can zip it up and upload it or add it to a git hub repository.

hideakitai, 2023-12-10:
Sorry, I must have misunderstood.

If the background color of the display changes, it must mean that some unexpected data is being transmitted. What kind of operation would change the background color that way?

I think you should check if the hardware connection is correct, because Slave will not send data if it does not receive data from Master.

navycuda, 2023-12-10:
Please forgive me as I'm still such a novice.

However I believe that the component that controls the tft-40p connector also communicates via SPI.  I think that's where the issue lies.  I just don't know how to find it.  Managing the instantiation order seems to have "solved" it though and I can successfully communicate via spi once the feather is running.

hideakitai, 2023-12-10:
I'm glad you were able to solve this, but it may be that the master (feather) needs to be initialized first. If you still need help, I need the wiring diagrams for these devices. If this is not a problem, I would appreciate it if you could close the issue.

navycuda, 2023-12-10:
Thank you so much for your time.  I don't want to take any more of it up for such an obscure issue that seems solved!

-Shane

closed 2023-12-10 by navycuda