hideakitai/ESP32SPISlave #31: begin/end

maarten-pennings, 2024-02-27:
I didn't want to pollute the other issue with this question.
Yes it is a question, maybe I should try it out myself.

As you might recall I have a system consisting of an ESP and a sensor.
The ESP triggers the sensor, and after a while the sensor masters one SPI message to the ESP.

Unfortunately, it gets more complex. There are actually two variants (A and B) of the sensor. 
Either one is connected to the ESP. 
And they would be connected to different pins of the ESP.
I would like to have auto-detect in the ESP firmware of the tow variants.

I was thinking of something like this.

```cpp
ESP32SPISlave spiin;
uint8_t spi_buf[SPI_BUF_SIZE];
void connect() {

  // Setup SPI slave controller for variant A
  spiin.setDataMode(SPI_MODE3); 
  spiin.begin(SPIIN_BUS, SPIIN_SCLK_A, -1, SPIIN_MOSI_A, SPIIN_SSEL_A);

  // Put a buffer in the reception queue if none is there
  if( spiin.hasTransactionsCompletedAndAllResultsHandled() ) {
    spiin.queue(NULL, spi_buf, SPI_BUF_SIZE);
    spiin.trigger();
  }

  // trigger sensor A
  ...

  // Wait till data is received from A
  uint32_t start = micros();
  while( ! spiin.hasTransactionsCompletedAndAllResultsReady(1)  &&  micros()-start<100 ) { /*wait*/ }

  // Did we receive data from A (or was there a timeout)?
  if( spiin.hasTransactionsCompletedAndAllResultsReady(1) ) {
    size_t num = spiin.numBytesReceived();
    ... use spi_buf[0..num-1] ...
  } else {
    // Timeout: no answer from A
    spiin.end();
    // Now try variant B
    spiin.setDataMode(SPI_MODE1); 
    spiin.begin(SPIIN_BUS, SPIIN_SCLK_B, -1, SPIIN_MOSI_B, SPIIN_SSEL_B);

    // Put a buffer in the reception queue if none is there
    if( spiin.hasTransactionsCompletedAndAllResultsHandled() ) {
      spiin.queue(NULL, spi_buf, SPI_BUF_SIZE);
      spiin.trigger();
    }
  
    // trigger sensor B
    ...
    // Wait till data is received from B
    ...   
    // Did we receive data from B (or was there a timeout)?
    ...
  }

}
```  

Does this work; the "pain" is in the `else` of the `if`.
- Does `end()` stop `spiin`?
- Does it stop the underlying task, does it free the `spiin-buf`?
- Can I do a `begin` with new mode and new pins?

hideakitai, 2024-02-27:
As you can see in the source code,

> Does end() stop spiin?

Yes

> Does it stop the underlying task, 

Yes

> does it free the spiin-buf?

No

> Can I do a begin with new mode and new pins

Yes

hideakitai, 2024-03-05:
If you still need help, please feel free to reopen this issue.

closed 2024-03-05 by hideakitai