hideakitai/ESP32SPISlave #34: Transfer in the background example

MichaelGallinger, 2024-02-28:
I'm using two ESP32 Devkit V1s with the "Transfer in the background example". There are two things I have been having issues with. I am new to using SPI so sorry if these have obvious answers.

The master constantly returns "successfully received expected data from slave" while the slave is stuck on "wait for the completion of the queued transactions..." and never prints the "successfully received expected data from master".

The second point I'm a bit confused about is where the data is being sent, are tx_buff and rf_buff the data being transfered? Where would I read the data to ensure it was verified before use?

I verified the pins as seen below, and changed the PIN_SS = 15 to PIN_SS = 5 however this did not affect the output.
MOSI GPIO Pin: 23
MISO GPIO Pin: 19
SCK GPIO Pin: 18
SS GPIO Pin: 5

hideakitai, 2024-02-28:
Could you give me a serial port output for both master and slave? Also, please enable debug logging mode by setting Arduino IDE > Tools > Core Debug Level > Debug.

> The second point I'm a bit confused about is where the data is being sent, are tx_buff and rf_buff the data being transfered? Where would I read the data to ensure it was verified before use?

Yes. tx_buf will be automatically sent, and rx_buf will be automatically updated if the transaction is completed. You can validate the completion of all the transactions by checking `hasTransactionsCompletedAndAllResultsReady()`.

hideakitai, 2024-02-29:
Are you using transfer_in_the_background_slave.ino example, “as is”? Or do you have any changes?

hideakitai, 2024-02-29:
If you use it without any modification, I can't understand what happens from your log. Could you give me an image captured with a logic analyzer or oscilloscope?

hideakitai, 2024-02-29:
Also, could you try other examples, “as is”?

Mike-G70, 2024-02-29:
Sorry, I don't have a logic analyzer or oscilloscope. I think I will use another communication method however, hopefully this may be useful. Please let me know if you want more information/tests done, thanks.

Here are the synced outputs as close to what I have:
Master (Output as is):
  13:27:31.648 -> successfully received expected data from slave
  13:27:33.610 -> successfully received expected data from slave
  13:27:35.628 -> successfully received expected data from slave
Slave (Output as is):
  13:27:35.189 -> start spi slave
  13:27:35.189 -> initialize tx/rx buffers
  13:27:35.189 -> execute transaction in the background
  13:27:35.189 -> [  2061][D][ESP32SPISlave.h:134] spi_slave_task(): [ESP32SPISlave] new transaction request received (size = 1)
  13:27:35.235 -> wait for the completion of the queued transactions...

The only change I made was making PIN_SS = 5 from 15, changing it back to 5 did not have an effect.

An additional weird effect I found was when the single esp32 master was not connected to anything it still outputs successfully received data most of the time:

Master output (No wired connections)
  13:32:33.822 -> successfully received expected data from slave
  13:32:35.779 -> successfully received expected data from slave
  13:32:37.813 -> successfully received expected data from slave
  13:32:39.783 -> master [0-26]: 00 01 02 03 04 05 06 07 44 0B 0B 01 88 FF FF FF 40 0C 0C 02 84 FF FF FF 48 0A 0A 
  13:32:39.831 -> slave [0-26]: FF FF FF FF FF FF FF FF 02 00 00 00 74 DC FB 3F 00 12 13 17 0F 00 00 00 01 10 4C 
  13:32:39.831 -> unexpected difference found between master/slave data
  13:32:41.817 -> successfully received expected data from slave

Additionally, when running many of the examples errors occur during compilation. I've tried reinstalling the Arduino IDE as well as the library but it did not affect the output. This is the "transfer_in_the_background_slave" code which compiles fine on my laptop but not my computer, both are running Arduino IDE 2.3.2

Transfer_in_the_background_master: No Errors

Transfer_in_the_background_slave Error:
```
In file included from C:\Users\Mike\AppData\Local\Temp\.arduinoIDE-unsaved2024129-11160-1a6u1vq.barz\transfer_in_the_background_slave\transfer_in_the_background_slave.ino:1:0:
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:39:34: error: 'function' in namespace 'std' does not name a template type
 using spi_slave_user_cb_t = std::function<void(spi_slave_transaction_t*, void*)>;
                                  ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:63:28: error: 'SOC_SPI_MAXIMUM_BUFFER_SIZE' was not declared in this scope
         .max_transfer_sz = SOC_SPI_MAXIMUM_BUFFER_SIZE,
                            ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:67:5: error: too many initializers for 'spi_bus_config_t'
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:68:29: error: 'SPI2_HOST' was not declared in this scope
     spi_host_device_t host {SPI2_HOST};
                             ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:68:38: error: cannot convert '<brace-enclosed initializer list>' to 'spi_host_device_t' in initialization
     spi_host_device_t host {SPI2_HOST};
                                      ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:81:9: error: 'spi_slave_user_cb_t' does not name a type
         spi_slave_user_cb_t user_cb;
         ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:85:9: error: 'spi_slave_user_cb_t' does not name a type
         spi_slave_user_cb_t user_cb;
         ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In function 'void arduino::esp32::spi::slave::spi_slave_post_setup_cb(spi_slave_transaction_t*)':
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:93:30: error: 'struct arduino::esp32::spi::slave::spi_slave_cb_user_context_t::<anonymous>' has no member named 'user_cb'
     if (user_ctx->post_setup.user_cb) {
                              ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:94:30: error: 'struct arduino::esp32::spi::slave::spi_slave_cb_user_context_t::<anonymous>' has no member named 'user_cb'
         user_ctx->post_setup.user_cb(trans, user_ctx->post_setup.user_arg);
                              ^
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In function 'void arduino::esp32::spi::slave::spi_slave_post_trans_cb(spi_slave_transaction_t*)':
[... cut ...]
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
C:\Users\Mike\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:572:66: error: 'to_string' is not a member of 'std'
         std::string task_name = std::string("spi_slave_task_") + std::to_string(this->ctx.if_cfg.spics_io_num);
                                                                  ^

exit status 1

Compilation error: exit status 1
```

hideakitai, 2024-02-29:
First, please confirm your environment.

- Please use the latest esp32 platform (v2.0.14)
- Please use the latest ESP32PISlave library (v0.6.0)
- Your connection seems to be unstable. Please refer https://github.com/hideakitai/ESP32SPISlave?tab=readme-ov-file#notes-for-communication-errors

Next, please make sure to start **slave first**. Then please start the master. Slave wait for the transaction from master, and then master start transaction. The slave receives and responds to the master's transaction.

hideakitai, 2024-03-08:
If you still need help, please feel free to reopen this issue.

closed 2024-03-08 by hideakitai