hideakitai/ESP32SPISlave #37: Massive compilation errors on 0.2.0 and 0.6.0

Frosty-STI, 2024-04-22:
Hi, 

On compiling an SPI slave script I got from https://microcontrollerslab.com/esp32-spi-communication-tutorial-arduino/#google_vignette, I receive these compilation errors:

In file included from C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino:1:
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:52:29: error: 'HSPI_HOST' was not declared in this scope; did you mean 'SPI3_HOST'?
   52 |     spi_host_device_t host {HSPI_HOST};
      |                             ^~~~~~~~~
      |                             SPI3_HOST
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:52:38: error: cannot convert '<brace-enclosed initializer list>' to 'spi_host_device_t' in initialization
   52 |     spi_host_device_t host {HSPI_HOST};
      |                                      ^
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::begin(uint8_t)':
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:59:43: error: 'VSPI' was not declared in this scope; did you mean 'SPI'?
   59 |         bus_cfg.mosi_io_num = (spi_bus == VSPI) ? MOSI : 13;
      |                                           ^~~~
      |                                           SPI
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:173:36: error: 'HSPI_HOST' was not declared in this scope; did you mean 'SPI3_HOST'?
  173 |         host = (spi_bus == HSPI) ? HSPI_HOST : VSPI_HOST;
      |                                    ^~~~~~~~~
      |                                    SPI3_HOST
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:173:48: error: 'VSPI_HOST' was not declared in this scope; did you mean 'SPI3_HOST'?
  173 |         host = (spi_bus == HSPI) ? HSPI_HOST : VSPI_HOST;
      |                                                ^~~~~~~~~
      |                                                SPI3_HOST

exit status 1

Compilation error: exit status 1

This is on version 0.2.0. I've tried 0.6.0 as well, and these are my compilation errors:

In file included from C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino:1:
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:615:14: error: 'string' is not a member of 'std'
  615 |         std::string task_name = std::string("spi_slave_task_") + std::to_string(this->ctx.if_cfg.spics_io_num);
      |              ^~~~~~
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:8:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
    7 | #include <driver/spi_slave.h>
  +++ |+#include <string>
    8 | #include <vector>
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:616:59: error: 'task_name' was not declared in this scope; did you mean 'basename'?
  616 |         int ret = xTaskCreatePinnedToCore(spi_slave_task, task_name.c_str(), SPI_SLAVE_TASK_STASCK_SIZE, static_cast<void*>(&this->ctx), SPI_SLAVE_TASK_PRIORITY, &spi_task_handle, 1);
      |                                                           ^~~~~~~~~
      |                                                           basename
C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino: In function 'void loop()':
C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino:49:15: error: no matching function for call to 'arduino::esp32::spi::slave::Slave::wait(uint8_t [32], uint8_t [32], const uint32_t&)'
   49 |     slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);
      |     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:378:25: note: candidate: 'std::vector<unsigned int> arduino::esp32::spi::slave::Slave::wait(uint32_t)'
  378 |     std::vector<size_t> wait(uint32_t timeout_ms = 0)
      |                         ^~~~
c:\Users\brend\OneDrive\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:378:25: note:   candidate expects 1 argument, 3 provided
C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino:55:18: error: 'using ESP32SPISlave = class arduino::esp32::spi::slave::Slave' {aka 'class arduino::esp32::spi::slave::Slave'} has no member named 'available'
   55 |     while (slave.available()) {
      |                  ^~~~~~~~~
C:\Users\brend\OneDrive\Documents\Arduino\LoRaSPISLAVE\LoRaSPISLAVE.ino:60:16: error: 'using ESP32SPISlave = class arduino::esp32::spi::slave::Slave' {aka 'class arduino::esp32::spi::slave::Slave'} has no member named 'pop'
   60 |          slave.pop();
      |                ^~~

exit status 1

Compilation error: no matching function for call to 'arduino::esp32::spi::slave::Slave::wait(uint8_t [32], uint8_t [32], const uint32_t&)'

This is my code:

#include <ESP32SPISlave.h>
#include <SPI.h>

ESP32SPISlave slave;

static constexpr uint32_t BUFFER_SIZE {32};
uint8_t spi_slave_tx_buf[BUFFER_SIZE];
uint8_t spi_slave_rx_buf[BUFFER_SIZE];

#define LED 2
void setup() {
    Serial.begin(115200);
    delay(2000);
    pinMode(LED, OUTPUT);
    // begin() after setting
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12 -> default
    // VSPI = CS:  5, CLK: 18, MOSI: 23, MISO: 19
    slave.setDataMode(SPI_MODE0);
    slave.begin();
    // slave.begin(VSPI);   // you can use VSPI like this

    // clear buffers
    memset(spi_slave_tx_buf, 0, BUFFER_SIZE);
    memset(spi_slave_rx_buf, 0, BUFFER_SIZE);
}

void loop() {
    // block until the transaction comes from master
    
    
    
    
    
    
    
    
    
    
    
    
    
    
    
    
    
    
    slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);

    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated
    char data;
    while (slave.available()) {
        // show received data
         Serial.print("Command Received: ");
         Serial.println(spi_slave_rx_buf[0]);
         data = spi_slave_rx_buf[0];
         slave.pop();
    }
    if(data == 1 )
    {
        Serial.println("Setting LED active HIGH ");
        digitalWrite(LED, HIGH);
    }
    else if(data == 0 )
    {
        Serial.println("Setting LED active LOW ");
        digitalWrite(LED, LOW);
    }
     Serial.println("");
}

What in the world is causing all of these errors?

hideakitai, 2024-04-23:
Please update the esp32 board to the latest v2.0.15 foram boards manager

Frosty-STI, 2024-04-23:
> Please update the esp32 board to the latest v2.0.15 foram boards manager

I have 🫠

hideakitai, 2024-04-23:
Please try the example of the latest releases of this library (v0.6.0) with the latest esp32 board (v2.0.15). Almost all the APIs have been changed between v0.2.0 and v0.6.0.

hideakitai, 2024-04-27:
If you still need help, please fell free to reopen this issue.

closed 2024-04-27 by hideakitai

Bennyggcc, 2025-01-08:
Hi, @hideakitai, I get the same error. I have used library (v0.6.0) and esp32 board (v2.0.15).
[image]
[image]

The error is "Compilation error: no matching function for call to 'arduino::esp32::spi::slave::Slave::wait(uint8_t [32], uint8_t [32], const uint32_t&)'". It seems that slave.wait function need a parameter "timeout_ms". How to solve this issue? Thank you!
[image]

Bennyggcc, 2025-01-09:
I turn to library (v0.2.0) and solve the problem.