hideakitai/ESP32SPISlave #38: [QUERY] Handling data integrity and acknowledgement

programmeddeath1, 2024-05-09:
I am trying to send multiple motor point messages from esp32  to slave esp32. I am using this library on the slave end to receive the data 
```
    memcpy(spi_master_tx_buf, point->mastercmd, 5);
    memcpy(spi_master_tx_buf + 5, point->pos, 3 * sizeof(float));
    memcpy(spi_master_tx_buf + 5 + 3 * sizeof(float), point->velo, 3 * sizeof(float));
    memcpy(spi_master_tx_buf + 5 + 6 * sizeof(float), point->accel, 3 * sizeof(float));
    memcpy(spi_master_tx_buf + 5 + 9 * sizeof(float), &point->vacc, sizeof(float));
    
    // Log buffer contents if needed
    logBuffer(spi_master_tx_buf, sizeof(spi_master_tx_buf));
  
    hspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
    digitalWrite(HSPI_SS, LOW);

    uint8_t ack = 0;
    for (int i = 0; i < sizeof(spi_master_tx_buf); i++) {
        hspi->transfer(spi_master_tx_buf[i]);  // Send data
    }
    ack = hspi->transfer(0);  // Send dummy byte to receive the ACK/NACK
    
    digitalWrite(HSPI_SS, HIGH);
    hspi->endTransaction();
```
Slave code -
```
void loop() {

   if (slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE)) {    
       // Deserialize data
       while (slave.available()) {
           // show received data
           Serial.println("Command Received: ");
           //  data = spi_slave_rx_buf[0];
           // Deserialize data from the SPI receive buffer to the `trajPoint` structure
           memcpy(point.mastercmd, spi_slave_rx_buf, 5);  // Adjusted size to 5
           memcpy(point.pos, spi_slave_rx_buf + 5, 3 * sizeof(float));
           memcpy(point.velo, spi_slave_rx_buf + 5 + 3 * sizeof(float), 3 * sizeof(float));
           memcpy(point.accel, spi_slave_rx_buf + 5 + 6 * sizeof(float), 3 * sizeof(float));
           memcpy(&point.vacc, spi_slave_rx_buf + 5 + 9 * sizeof(float), sizeof(float));

           // Print the received data
           Serial.print("Command: ");
           Serial.println(point.mastercmd);  // Should print "EXEC"
           Serial.print("Positions: ");
           Serial.print(point.pos[0]);
           Serial.print(", ");
           Serial.print(point.pos[1]);
....
#Remaining print
           slave.pop();
       }       
   }
} 
```
When i check the data on the slave end, it seems to drop a message after every 3 or 4 points, and the message drop seems random, it just shows all zeroes on the slave end. I logged the master end before the transfer, even the byte data after serialization is proper. I tried sending messages after 1 second and still the messages after a random interval get dropped. Hers the sample data - 
```
  Command Received: 
  Command: EXEC
  Positions: 133.45, 133.45, 216.19
  Velocities: 500.00, 500.00, 338.35
  Accelerations: 0.00, 0.00, 0.00
  Vacc: 92.00
  Raw buffer content:
  45 58 45 43 0 7 EE D 43 7 EE D 43 D3 F2 5D 
  43 6F 60 EE 43 6F 60 EE 43 9 2D A9 43 0 C0 DA 
  C5 0 C0 DA C5 0 0 0 0 0 0 BA 42 0 0 0 
  Command Received: 
  Command: EXEC
  Positions: 141.93, 141.93, 221.95
  Velocities: 476.75, 476.75, 338.35
  Accelerations: -7000.00, -7000.00, 0.00
  Vacc: 93.00
  Raw buffer content:
  45 58 45 43 0 61 8 15 43 61 8 15 43 AA B5 63 
  43 53 C8 B2 43 53 C8 B2 43 9 2D A9 43 0 C0 DA 
  C5 0 C0 DA C5 0 0 0 0 0 0 BC 42 0 0 0 
  Command Received: 
  Command: EXEC
  Positions: 149.03, 149.03, 227.71
  Velocities: 357.57, 357.57, 338.35
  Accelerations: -7000.00, -7000.00, 0.00
  Vacc: 94.00
  Raw buffer content:
  45 58 45 43 0 33 1B 1A 43 33 1B 1A 43 BC C1 68 
  43 6F 60 6E 43 6F 60 6E 43 6F 60 6E 43 0 C0 DA 
  C5 0 C0 DA C5 0 C0 DA C5 0 0 BE 42 0 0 0 
  Command Received: 
  Command: EXEC
  Positions: 154.11, 154.11, 232.76
  Velocities: 238.38, 238.38, 238.38
  Accelerations: -7000.00, -7000.00, -7000.00
  Vacc: 95.00
  Raw buffer content:
  45 58 45 43 0 7E 26 1D 43 7E 26 1D 43 7 CD 6B 
  43 6F 60 EE 42 6F 60 EE 42 6F 60 EE 42 0 C0 DA 
  C5 0 C0 DA C5 0 C0 DA C5 0 0 C0 42 0 0 0 
  Command Received: 
  Command: EXEC
  Positions: 157.15, 157.15, 235.80
  Velocities: 119.19, 119.19, 119.19
  Accelerations: -7000.00, -7000.00, -7000.00
  Vacc: 96.00
  Raw buffer content:
  45 58 45 43 0 42 2A 1E 43 42 2A 1E 43 CB D0 6C 
  43 0 0 0 0 0 0 0 0 0 0 7A A9 0 0 0 
  0 0 0 0 0 0 0 0 0 0 0 C2 42 0 0 0 
  Command Received: 
  Command: EXEC
  Positions: 158.17, 158.17, 236.82
  Velocities: 0.00, 0.00, -0.00
  Accelerations: 0.00, 0.00, 0.00
  Vacc: 97.00
  Raw buffer content:
  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
  Command Received: 
  Command: 
  Positions: 0.00, 0.00, 0.00
  Velocities: 0.00, 0.00, 0.00
  Accelerations: 0.00, 0.00, 0.00
  Vacc: 0.00
  Raw buffer content:
  24 27 A6 A2 80 0 0 0 0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
  Command Received: 
  Command: $'⸮⸮⸮
  Positions: 0.00, 0.00, 0.00
  Velocities: 0.00, 0.00, 0.00
  Accelerations: 0.00, 0.00, 0.00
  Vacc: 0.00
```

What could be causing this random message drops? 
Is there a  way to implement ack nack such that if the slave doesn't receive the proper msg it sends a NACK and the master can try to send the same point once more.

I checked the transfer time, and it is taking around 3 to 4 ms, hence even a transfer time of 7 to 8ms after ack nack every 4th or 5th message is still tolerable, not okay but workable.

hideakitai, 2024-05-19:
I'm sorry for not getting back to you sooner. How about hardware noise? Please refer to the README https://github.com/hideakitai/ESP32SPISlave?tab=readme-ov-file#notes-for-communication-errors

hideakitai, 2024-06-02:
If you still need help, please reopen this issue.

closed 2024-06-02 by hideakitai