hideakitai/ESP32SPISlave #40: Error in driver when using  board manager esp32 v3.0.1

maarten-pennings, 2024-06-13:
Espressif has released their board support package for esp32 in the 2.x line up to 2.0.17.
But they recently switched to 3.0.1, the switch in major version number indicating major changes.

Unfortunately, there are now compiler errors in the ESP32SPIslave driver:

```txt
In file included from C:\Users\maarten\AppData\Local\Temp\.arduinoIDE-unsaved2024513-18580-4anhiu.ews71\transfer_one_by_one_slave\transfer_one_by_one_slave.ino:1:
c:\Users\maarten\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:70:5: warning: missing initializer for member 'spi_bus_config_t::isr_cpu_id' [-Wmissing-field-initializers]
   70 |     };
      |     ^
c:\Users\maarten\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h: In member function 'bool arduino::esp32::spi::slave::Slave::initialize(uint8_t)':
c:\Users\maarten\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:615:14: error: 'string' is not a member of 'std'
  615 |         std::string task_name = std::string("spi_slave_task_") + std::to_string(this->ctx.if_cfg.spics_io_num);
      |              ^~~~~~
c:\Users\maarten\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:8:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
    7 | #include <driver/spi_slave.h>
  +++ |+#include <string>
    8 | #include <vector>
c:\Users\maarten\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:616:59: error: 'task_name' was not declared in this scope; did you mean 'basename'?
  616 |         int ret = xTaskCreatePinnedToCore(spi_slave_task, task_name.c_str(), SPI_SLAVE_TASK_STASCK_SIZE, static_cast<void*>(&this->ctx), SPI_SLAVE_TASK_PRIORITY, &spi_task_handle, 1);
      |                                                           ^~~~~~~~~
      |                                                           basename

exit status 1

Compilation error: exit status 1
```

Quick analysis
- `missing initializer for member 'spi_bus_config_t::isr_cpu_id'`  
  suggests a field has been added to a base struct; suggest to replace the comment on line 69 by
  `.isr_cpu_id = INTR_CPU_ID_AUTO,` ???
- `error: 'string' is not a member of 'std'`  
  suggests a missing header file; add `#include <string>` to top of file
- `error: 'task_name' was not declared in this scope;`  
  is probably caused by the previous error

hideakitai, 2024-06-13:
It has already been fixed in v0.6.1. Please try

maarten-pennings, 2024-06-14:
The first compilation problem listed above is not yet fixed.

I compiled `transfer_one_by_one_slave` using board support package 'esp32 by espressif' 3.0.1:

```
In file included from C:\Users\mpen\AppData\Local\Temp\.arduinoIDE-unsaved2024514-7720-1bt2agz.6q0ul\transfer_one_by_one_slave\transfer_one_by_one_slave.ino:1:
c:\Users\mpen\Documents\Arduino\libraries\ESP32SPISlave/ESP32SPISlave.h:71:5: warning: missing initializer for member 'spi_bus_config_t::isr_cpu_id' [-Wmissing-field-initializers]
   71 |     };
      |     ^
```

The field `.isr_cpu_id` from the struct `spi_bus_config_t` still needs an initializer.

hideakitai, 2024-06-14:
It’s a warning, but can’t you compile? (in CI and my env compiles successfully)

maarten-pennings, 2024-06-14:
Yes, it is a "only" warning, but I need my project to compile clean.

I believe this 3.x Arduino board support package from espressif has a newer c compiler and/or more stringent settings. 

Although this is a warning, I think it should be fixed. The compiler complains that one field of a (inherited) struct is not initialized. This sounds like the "base class" works with uninitialized data. It's just one extra line of code.

hideakitai, 2024-06-15:
If you want to keep your project clean, I recommend sending me a pull request or creating your own ;)

closed 2024-06-15 by hideakitai

maarten-pennings, 2024-06-25:
I'm back from vacation, and wanted to make a pull request, but it compiles clean now. Looking at your fix, you clearly know much more than me about this stuff. Thanks for solving it.