hideakitai/ESP32SPISlave #44: How to use with Adafruit Qualia board with ESP32-S3-WROOM1 module

karunt, 2024-08-05:
New to using non-ESP boards with ESP modules. I understand that default pin numbers for ESP32-S3 SPI2 bus are 10(CS), 11(MOSI), 12(SCK) and 13 (MISO). I'm using the Adafruit Qualia board which hosts the ESP32-S3-WROOM1 module (which uses the ESP32-S3 chip). From the ESP32-S3-WROOM1 module specs, I gather that the module's pin numbers which related to SPI2 pins on the ESP32-S3 chip are 18(CS), 19(MOSI), 20(SCK) and 21(MISO).

So when I use slave.begin() without any arguments, if I understand the library correctly, it defaults to the default SPI2 bus pins on the ESP32-S3 chip. But when using the Adafruit Qualia board with the ESP32-S3-WROOM1 module, do I continue to use these default SPI2 bus pins, or do I have to specify my pin numbers in slave.begin() to match corresponding SPI pins on the ESP32-S3-WROOM1 module (pins 18, 19, 20 and 21) instead?

Thanks.

hideakitai, 2024-08-05:
You can use any pins. I think it is easy to understand to use ESP32-S3-WROOM1 pins. Please take a look at this document for more details.

https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32s3/api-reference/peripherals/spi_slave.html#gpio-matrix-and-io-mux

karunt, 2024-08-06:
Thanks. I guess what's confusing me is which pin numbers I should use in slave.begin(). For instance, on the Qualia board, MOSI and MISO are connected to IO7 and IO6, respectively, on the ESP32-S3-WROOM1 module (which translates to pin #s 7 and 6, respectively, on the WROOM1 module (Table 3: Pin Definitions in [https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf]).  However, the same IO7 and IO6 map to pins 12 and 11, respectively, on the ESP32-S3 chip (Table 2-3 in [https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf]). So which pin numbers should I use in slave.begin()?

hideakitai, 2024-08-14:
I doubt I understand it correctly, but please use the `XX` number with the name `GPIOXX`.

The pin numbers without the `GPIO` prefix are the physical pin numbers of each package. For example, they are physically numbered counterclockwise, starting from the upper left corner.

hideakitai, 2024-08-19:
If you still need help, please reopen this issue.

closed 2024-08-19 by hideakitai