hideakitai/ESP32SPISlave #47: Replicate Arduino Slave behaviour

gbkwiatt, 2024-10-12:
I am pulling my hair right now, and wonder if anyone can assist me.  I am trying to replicate behaviour from Arduino code as follow 
```cpp
void setup() {
  pinMode(MISO, OUTPUT); //Pin 12
  pinMode(SS, INPUT);  //Pin 10
  pinMode(0, INPUT_PULLUP); //Button, default state high.
  //intitialize pull counter
  pullcount = 0; 

  // turn on SPI in slave mode
  SPCR |= _BV(SPE);

  // turn on interrupts
  SPCR |= _BV(SPIE);
  //interrupt routine on pin 2, calls resetCount function on falling edge of SS signal
  //without this call, the analog data doesn't initialize in the proper order
  //RISING has been required for some users, investigation ongoing
  attachInterrupt(0, resetCount, FALLING); 
}

// reset the pull counter when called (interrupt)
void resetCount()
{
  pullcount = 0; 
  SPDR = buttonbit1; //byte 1 buttons
}

// SPI interrupt routine
//This interrupt is called everytime SPI transfer is called from the master
ISR (SPI_STC_vect)
{
 if (pullcount == 0) // byte 2 buttons
    {
      SPDR = buttonbit2; 
      pullcount++; 
    }
    else if (pullcount == 1) //byte 3 buttons
    {
      SPDR = 255;
      pullcount++; 
    }
    //... 
}

void loop()
{
 // just set the data to send
}
```

hideakitai, 2024-10-22:
If you are not using my library, I suggest you ask on the Arduino Forum.

closed 2024-10-22 by hideakitai

hideakitai, 2024-10-28:
The use of interrupts in this issue is not related to the contents of my library (and [spi slave driver](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/peripherals/spi_slave.html)) but to the use of interrupts in general. I wonder if adding interrupt functionality to pins with SPI assigned to them is possible, so I suggest asking on the Arduino Forum.

gbkwiatt, 2024-10-28:
> The use of interrupts in this issue is not related to the contents of my library (and [spi slave [...]

I think you're missing the point. The interrupt is the SPI interrupt 

SPI_STC_vect is a Serial Transfer Complete interrupt vector

Which I believe has equivalent in esp32 slave driver, and that was the question about. So it is related to your library and it is related to esp32 slave library.

I think `post_setup_cb` and `post_trans_cb` are those interrupts but I am not sure how to use it in you're library.

hideakitai, 2024-10-28:
> SPI_STC_vect is a Serial Transfer Complete interrupt vector

I think what you'd like to do is not covered by the ESP-IDF's SPI slave driver. Because `post_setup_cb` is just a `Callback called after the SPI registers are loaded with new data.` and `post_trans_cb` is a `Callback called after a transaction is done`.

https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/peripherals/spi_slave.html#_CPPv4N28spi_slave_interface_config_t13post_setup_cbE

The lower-level interrupts are not supported in this library, and I'm unfamiliar with them. So, I suggested asking on the Arduino Forum.

> I think post_setup_cb and post_trans_cb are those interrupts but I am not sure how to use it in you're library.

Did you see this example?

https://github.com/hideakitai/ESP32SPISlave/blob/main/examples/transfer_in_the_background_with_cb/transfer_in_the_background_with_cb_slave/transfer_in_the_background_with_cb_slave.ino