hideakitai/ESP32SPISlave #48: Crash while using WiFi

SwapnilPande, 2024-10-30:
Hello!

I've been trying to setup an ESP32-S3 as an SPI Slave while using WiFi with a Raspberry Pi as the master. I am able to get stable communications at 5 MHz with WiFi disable. However, the driver seems to crash if I try to enable WiFi as well. I have tried to use DMA/non-DMA, different SPI modes with no success. The only thing that seems to improve the stability is dropping the SPI clock frequency significantly (500 Hz). 

I will try to provide a minimal working example soon. Do you have thoughts as to what might be causing this?

SwapnilPande, 2024-10-30:
```
#include <ESP32SPISlave.h>
#include <WiFi.h>

#define MOSI 35
#define MISO 48
#define SCK 47
#define SS 21

uint8_t tx_buf[64]{0};
uint8_t rx_buf[64]{0};

ESP32SPISlave slave;

void setup()
{
  Serial.begin(115200);

  delay(4000);

  // Connect to Wi-Fi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.println("Connecting to WiFi...");
  }

  Serial.println("Connected to WiFi");

  slave.setDataMode(SPI_MODE3);  // default: SPI_MODE0
  slave.setQueueSize(1);         // default: 1

  slave.begin(FSPI, SCK, MISO, MOSI, SS);
}

void loop()
{
  //   Allocate buffers
  const size_t received_bytes = slave.transfer(tx_buf, rx_buf, 64, 50);
}

```

Here is the minimum example

hideakitai, 2024-10-31:
The driver and the program work fine if it works with a lower frequency. I suspect the increased power consumption by WiFi causes the noise on the SPI lines. The noise prevents the higher frequency. Could you take a look at the SPI line waveforms?

hideakitai, 2024-11-10:
If you still need help, please reopen this issue.

closed 2024-11-10 by hideakitai

SwapnilPande, 2024-12-12:
Thank you for the quick response, I've finally gotten around to try to debug this. Crashes are always caused by the below error:

```
[ 63003][E][ESP32DMASPISlave.h:169] spi_slave_task(): [ESP32DMASPISlave] failed to execute spi_slave_get_trans_result(): 0x107
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
```

I will try to break out my scope to look at the waveforms today