hideakitai/ESP32SPISlave #51: Lib no longer compiles in Arduino with board 'esp32 by Espressif Systems' version '3.1.1'

maarten-pennings, 2025-01-24:
With an update to the latest greatest version `3.1.1` of the board `esp32 by Espressif Systems`, the library no longer compiles.

[image]

I tried the example `transfer_in_the_background`. It comes with the error

```
ESP32SPISlave/ESP32SPISlave.h:70:23: error: 'INTR_CPU_ID_AUTO' was not declared in this scope
```

[image]

I must say that the elif looks suspicious, it will never hit because 5.2 is already true for the first if.

```
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 1, 0)
        .isr_cpu_id = INTR_CPU_ID_AUTO,
#elif ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 2, 0)
        .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
#endif
```

Thanks

DrSimgear, 2025-01-28:
I am also having the same issue

closed 2025-02-02 by hideakitai

hideakitai, 2025-02-02:
Fixed by [v0.6.5](https://github.com/hideakitai/ESP32SPISlave/releases/tag/v0.6.5)