hideakitai/ESP32SPISlave #54: Wrong version in library.properties

maarten-pennings, 2025-02-04:
When downloading release `ESP32SPISlave-0.6.5.zip` and reading file `library.properties`, these are the contents:

```
name=ESP32SPISlave
version=0.6.3
author=hideakitai
maintainer=hideakitai
sentence=SPI Slave library for ESP32
paragraph=SPI Slave library for ESP32
category=Communication
url=https://github.com/hideakitai
architectures=*
```

Note that the `version` is not updated; it is still `0.6.3`. 
This will confuse Arduino IDE with its update policy.

Strangly enough, the [main branch its library.properties](https://github.com/hideakitai/ESP32SPISlave/blob/main/library.properties) has the correct version listed.

hideakitai, 2025-02-05:
Thank you, fixed https://github.com/hideakitai/ESP32SPISlave/releases/tag/v0.6.5

closed 2025-02-05 by hideakitai

maarten-pennings, 2025-02-06:
Also thank you, for the fast response.