hideakitai/ESP32SPISlave #55: ESP32-S3 empty bytes sent

dexter323i, 2025-03-18:
Hi!

I want to use ESP32-S3 to send 3 bytes towards an SPI master. Clock is 1Mhz, I have a logic analyzer, and no matter how I try, I cannot send anything to the master. Analyser says 0 was sent. I tried your non blocking transaction example, and it says hasTransactionsCompletedAndAllResultsHandled() is true, but received_bytes is 0.

What am I doing wrong?

```
#include <ESP32SPISlave.h>
#include "helper.h"

#define PIN_SPI_SS      10
#define PIN_SPI_SCK     12
#define PIN_SPI_MOSI    11
#define PIN_SPI_MISO    13

ESP32SPISlave spiSlave;
static constexpr size_t BUFFER_SIZE = 3;
static constexpr size_t QUEUE_SIZE = 1;
uint8_t tx_buf[BUFFER_SIZE] {2, 4, 8};
uint8_t rx_buf[BUFFER_SIZE] {2, 4, 8};

void setup() {
        Serial.begin(115200);

        delay(2000);

        spiSlave.setDataMode(SPI_MODE2);   
        spiSlave.setQueueSize(QUEUE_SIZE);
    
        spiSlave.begin();
}

void loop() {
           if (spiSlave.hasTransactionsCompletedAndAllResultsHandled()) {
                tx_buf[0] = 2;
                tx_buf[1] = 4;
                tx_buf[2] = 8;
                rx_buf[0] = 2;
                rx_buf[1] = 4;
                rx_buf[2] = 8;
                const size_t received_bytes = spiSlave.transfer(tx_buf, rx_buf, BUFFER_SIZE);

                Serial.println(received_bytes);
         }
}

```

Serial results:
0
0
0
0
0
0
0
0
0

Even when the ESP32 is powered. It is not even connected to the Master.

[image]

dexter323i, 2025-03-19:
I spent hours to find the issue. No luck. :( Current code:

```
#include <ESP32SPISlave.h>
#include <helper.h>

ESP32SPISlave spiSlave;
static constexpr size_t BUFFER_SIZE = 3;
static constexpr size_t QUEUE_SIZE = 1;
uint8_t tx_buf[BUFFER_SIZE] {2, 4, 8};
uint8_t rx_buf[BUFFER_SIZE] {0, 0, 0};

#define PIN_SPI_SS      10
#define PIN_SPI_SCK     12
#define PIN_SPI_MOSI    11
#define PIN_SPI_MISO    13

void setup() {
        pinMode(PIN_SPI_MISO, OUTPUT);
        pinMode(PIN_SPI_MOSI, INPUT_PULLDOWN);
        pinMode(PIN_SPI_SCK, INPUT);
        pinMode(PIN_SPI_SS, INPUT);

        Serial.begin(115200);

        delay(2000);

        spiSlave.setDataMode(SPI_MODE2);   
        spiSlave.setQueueSize(QUEUE_SIZE);
    
        spiSlave.begin(HSPI, PIN_SPI_SCK, PIN_SPI_MISO, PIN_SPI_MOSI, PIN_SPI_SS);
}

void loop() {
        tx_buf[0] = 2;
        tx_buf[1] = 4;
        tx_buf[2] = 8;
    
        spiSlave.queue(tx_buf, rx_buf, BUFFER_SIZE);
        spiSlave.trigger();
}
```

I left the MISO pin not connected to master device, to see if the slave ESP32-S3 even actuates it. It does: pulls low at the begining of the transmisson, then pulls high at the end, but no bytes sent.

[image]

[image]

[image]

[image]

Gartosz, 2025-03-30:
Hi, I'm facing similar issue on the ESP32-S3 too. Using the transfer_one_by_one example the slave has only zeros in the rx buffer and on the master some random bytes occur. I had to made some changes, as my master is Arduino Mega 2560, but other than that, it's the same. Changing baud doesn't help.

UltimumTech, 2025-04-18:
Hi, I made this as a starter guide for you to try out. I tested everything and it all works. Everything is commented for easy understanding. Thanks!

```
/**
 * ESP32 SPI Slave Example
 * 
 * This program configures an ESP32 microcontroller as an SPI slave device,
 * which can communicate with an SPI master device (like another microcontroller).
 */

// Include the ESP32 SPI Slave library
#include <ESP32SPISlave.h>

// Debugging configuration
#define DEBUG  // When defined, outputs debug information via serial (note: can cause SPI output glitches)

// Operation mode selection (choose one)
//#define SSActivated    // Faster mode that triggers on SS (Slave Select) pin changes
#define LogicActivated   // Slower mode that uses program logic for triggering

// Create a pointer for the SPI slave object
ESP32SPISlave *hspi = NULL;

// Define the pins used for SPI communication
#define HSPI_SDO 11   // SDO (MISO - Master In Slave Out) pin
#define HSPI_SCLK 12  // Serial Clock pin
#define HSPI_SS 10    // SS (Slave Select) pin

//-------------------------------------------------------------------
unsigned long ct;  // Variable to store current time

#ifdef SSActivated
// For SS-activated mode, we use an interrupt
volatile bool CSactivated;  // Atomic flag to indicate Chip Select state change

// Interrupt Service Routine that runs when the SS pin goes HIGH (transaction end)
void IRAM_ATTR handleCSrising() {
  CSactivated = true;  // Mark that a transaction has ended
}
#endif

//-------------------------------------------------------------------
// Define buffer sizes and data structures
static constexpr size_t BUFFER_SIZE = 8;  // Size of data buffer (8 bytes)
static constexpr size_t QUEUE_SIZE = 1;   // Size of transaction queue

// Transmit buffer - data to send to the master (initialized with values 1-8)
uint8_t tx_buf[BUFFER_SIZE]{ 1, 2, 3, 4, 5, 6, 7, 8 };

// Receive buffer is commented out (not used in this example)
//uint8_t rx_buf[BUFFER_SIZE] {0, 0, 0, 0, 0, 0, 0, 0};

//-------------------------------------------------------------------
void setup() {
  // Initialize serial communication at 115200 baud rate
  Serial.begin(115200);
  
  // Create and configure the SPI slave instance
  hspi = new ESP32SPISlave();
  hspi->setDataMode(SPI_MODE1);     // Set SPI mode (but then overridden below)
  hspi->setDataMode(SPI_MODE2);     // SPI_MODE2: CPOL=1, CPHA=0 (clock idle high, sample on falling edge)
  hspi->setQueueSize(QUEUE_SIZE);   // Set the size of the transaction queue
  
  // Initialize the SPI slave on HSPI bus with the specified pins
  // Parameters: SPI bus, clock pin, data out pin, data in pin (not used so -1), slave select pin
  hspi->begin(HSPI, HSPI_SCLK, HSPI_SDO, -1, HSPI_SS);
  
  #ifdef SSActivated
  // If using SS-activated mode, attach an interrupt to the SS pin
  attachInterrupt(HSPI_SS, handleCSrising, RISING);  // Call handleCSrising when SS pin goes from LOW to HIGH
[... 72 lines cut ...]
  tx_buf[2] = 3;
  tx_buf[3] = 4;
  tx_buf[4] = 5;
  tx_buf[5] = 6;
  tx_buf[6] = 7;
  tx_buf[7] = 8;
}

//-------------------------------------------------------------------
/**
 * Debug function to print buffer contents to serial monitor
 * Only compiled if DEBUG is defined
 */
#ifdef DEBUG
void debug_buffer() {
  Serial.println("");  // New line
  for (int i = 0; i < BUFFER_SIZE; i++) {
    Serial.print(" Byte ");
    Serial.print(i);
    Serial.print(": 0x");
    if (tx_buf[i] < 0x10) Serial.print('0');  // Add leading zero for values < 16
    Serial.print(tx_buf[i], HEX);             // Print the value in hexadecimal
  }
}
#endif ```

hideakitai, 2025-04-30:
Please try the examples "as is" first. Then you can modify as you want.

closed 2025-06-01 by hideakitai

Gartosz, 2025-06-20:
I was, but only on slave, as my master was Arduino Mega 2560, so I couldn't compile it even if I wanted to. The issue was fixed by adding 330 Ω resistors on MISO and MOSI lines.