hideakitai/ESP32SPISlave #57: Compilation breaks (in Arduino) with latest board package (3.2.0)

maarten-pennings, 2025-04-08:
When compiling in Arduino IDE 2.3.5, with the ESP32 package 3.2.0

[image]

where I have highest warning level

[image]

this error is reported

```
In file included from c:\Users\mpen\Documents\Arduino\libraries\OSP_aospi\src\aospi.cpp:177:
c:\Users\maarten\Documents\Arduino\libraries\OSP_aospi\src\slave/ESP32SPISlave.h:75:5: warning: missing initializer for member 'spi_bus_config_t::data_io_default_level' [-Wmissing-field-initializers]
   75 |     };
      |     ^
```

It seems that Espressif added `data_io_default_level` to `spi_bus_config_t`.

closed 2025-04-29 by hideakitai

hideakitai, 2025-04-29:
Fixed by https://github.com/hideakitai/ESP32SPISlave/pull/58

hideakitai, 2025-05-01:
Thank you, fixed in v0.6.7 https://github.com/hideakitai/ESP32SPISlave/pull/60